vsergeev/c-periphery #7: BeagleBone Black + Yocto not working

bernardoaraujor, 2017-01-17:
GPIO test failed at test_loopback.

Built with yocto.

vsergeev, 2017-01-26:
Did you have a physical loopback installed between the two GPIOs?

bernardoaraujor, 2017-01-26:
yeap

vsergeev, 2017-01-26:
Could you provide the log of the run?

bernardoaraujor, 2017-01-26:
Picture of the physical loopback: https://goo.gl/photos/1h5nGFbwL6nCPvdW6

Here's the log of the whole process, including steps used for Yocto's devtool:

bernardo@bernardo-ThinkCentre-Edge72:~/yocto/bbb$ devtool add ~/c-periphery/
NOTE: Creating workspace layer in /home/bernardo/yocto/bbb/workspace
NOTE: Enabling workspace layer in bblayers.conf
NOTE: Using source tree as build directory since that would be the default for this recipe
NOTE: Recipe /home/bernardo/yocto/bbb/workspace/recipes/c-periphery/c-periphery_git.bb has been automatically created; further editing may be required to make it fully functional
bernardo@bernardo-ThinkCentre-Edge72:~/yocto/bbb$ cat conf/local.conf 
#
# This file is your local configuration file and is where all local user settings
# are placed. The comments in this file give some guide to the options a new user
# to the system might want to change but pretty much any configuration option can
# be set in this file. More adventurous users can look at local.conf.extended
# which contains other examples of configuration which can be placed in this file
# but new users likely won't need any of them initially.
#
# Lines starting with the '#' character are commented out and in some cases the
# default values are provided as comments to show people example syntax. Enabling
# the option is a question of removing the # character and making any change to the
# variable as required.

#
# Machine Selection
#
# You need to select a specific machine to target the build with. There are a selection
# of emulated machines available which can boot and run in the QEMU emulator:
#
#MACHINE ?= "qemuarm"
#MACHINE ?= "qemuarm64"
#MACHINE ?= "qemumips"
#MACHINE ?= "qemumips64"
#MACHINE ?= "qemuppc"
#MACHINE ?= "qemux86"
#MACHINE ?= "qemux86-64"
#
# There are also the following hardware board target machines included for 
# demonstration purposes:
#
#MACHINE ?= "beaglebone"
#MACHINE ?= "genericx86"
#MACHINE ?= "genericx86-64"
#MACHINE ?= "mpc8315e-rdb"
#MACHINE ?= "edgerouter"
#
# This sets the default machine to be qemux86 if no other machine is selected:
MACHINE ??= "qemux86"

#
# Where to place downloads
#
# During a first build the system will download many different source code tarballs
# from various upstream projects. This can take a while, particularly if your network
# connection is slow. These are all stored in DL_DIR. When wiping and rebuilding you
# can preserve this directory to speed up this part of subsequent builds. This directory
# is safe to share between multiple builds on the same machine too.
#
# The default is a downloads directory under TOPDIR which is the build directory.
#
#DL_DIR ?= "${TOPDIR}/downloads"

#
# Where to place shared-state files
#
# BitBake has the capability to accelerate builds based on previously built output.
# This is done using "shared state" files which can be thought of as cache objects
# and this option determines where those files are placed.
#
# You can wipe out TMPDIR leaving this directory intact and the build would regenerate
[... 303 lines cut ...]
 [ OK ]  tests/test_gpio.c test_open_config_close():90  gpio_get_edge(&gpio, &edge) == 0
 [ OK ]  tests/test_gpio.c test_open_config_close():91  edge == GPIO_EDGE_BOTH
 [ OK ]  tests/test_gpio.c test_open_config_close():93  gpio_set_edge(&gpio, GPIO_EDGE_NONE) == 0
 [ OK ]  tests/test_gpio.c test_open_config_close():94  gpio_get_edge(&gpio, &edge) == 0
 [ OK ]  tests/test_gpio.c test_open_config_close():95  edge == GPIO_EDGE_NONE
 [ OK ]  tests/test_gpio.c test_open_config_close():98  gpio_close(&gpio) == 0
 [ OK ]  tests/test_gpio.c test_open_config_close():101  gpio_open(&gpio, pin_output, GPIO_DIR_OUT) == 0
 [ OK ]  tests/test_gpio.c test_open_config_close():103  gpio_close(&gpio) == 0
 [ OK ]  tests/test_gpio.c test_open_config_close():105  gpio_open(&gpio, pin_output, GPIO_DIR_PRESERVE) == 0
 [ OK ]  tests/test_gpio.c test_open_config_close():107  gpio_get_direction(&gpio, &direction) == 0
 [ OK ]  tests/test_gpio.c test_open_config_close():108  direction == GPIO_DIR_OUT
 [ OK ]  tests/test_gpio.c test_open_config_close():111  gpio_close(&gpio) == 0
 [ OK ]  Open/close test passed.

Starting test test_loopback():118
 [ OK ]  tests/test_gpio.c test_loopback():121  gpio_open(&gpio_in, pin_input, GPIO_DIR_IN) == 0
 [ OK ]  tests/test_gpio.c test_loopback():122  gpio_open(&gpio_out, pin_output, GPIO_DIR_OUT) == 0
 [ OK ]  tests/test_gpio.c test_loopback():125  gpio_write(&gpio_out, false) == 0
 [ OK ]  tests/test_gpio.c test_loopback():126  gpio_read(&gpio_in, &value) == 0
 [ OK ]  tests/test_gpio.c test_loopback():127  value == false
 [ OK ]  tests/test_gpio.c test_loopback():130  gpio_write(&gpio_out, true) == 0
 [ OK ]  tests/test_gpio.c test_loopback():131  gpio_read(&gpio_in, &value) == 0
 [FAIL]  tests/test_gpio.c test_loopback():132  value == true
test_gpio: tests/test_gpio.c:132: test_loopback: Assertion `r' failed.
Aborted

bernardoaraujor, 2017-01-26:
oops.. forgot to cat workspace/recipes/c-periphery/c-periphery_git.bb

I'll do it in a minute.

vsergeev, 2017-01-26:
I bet your GPIO numbers are somehow mapped differently. What kernel are you using?

Surest way to check though is with a multimeter on GPIO 38 (P8.3) and manually writing 1 or 0 `/sys/class/gpio/gpio38/value`, after writing `out` to `/sys/class/gpio/gpio38/direction`.

bernardoaraujor, 2017-01-26:
Indeed. Reproduced the commands, but the voltimeter reads only 3.3v, even after echoing 0.

vsergeev, 2017-03-22:
I've been running Arch Linux ARM, which is currently on 4.10.4. But the Linux package is the linux-am33x variant, which may have the GPIO mapping expected by the test.

vsergeev, 2017-03-22:
I'm going to close this issue for now, as it's just a platform configuration issue, rather than a library issue. But feel free to follow up here on the GPIO mapping.

closed 2017-03-22 by vsergeev