vsergeev/c-periphery #23: Support GPIO line bias

sgjava, 2020-05-02:
This is for pull up, pull down, etc. https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/tree/lib/core.c#n374

vsergeev, 2020-05-19:
I'm thinking:

``` diff
diff --git a/src/gpio.h b/src/gpio.h
index d489cec..eda6ca8 100644
--- a/src/gpio.h
+++ b/src/gpio.h
@@ -41,12 +41,27 @@ typedef enum gpio_edge {
     GPIO_EDGE_BOTH      /* Both edges X -> !X */
 } gpio_edge_t;
 
+typedef enum gpio_bias {
+    GPIO_BIAS_DEFAULT,      /* Default line bias */
+    GPIO_BIAS_PULL_UP,      /* Pull-up */
+    GPIO_BIAS_PULL_DOWN,    /* Pull-down */
+    GPIO_BIAS_DISABLE,      /* Disable line bias */
+} gpio_bias_t;
+
+typedef enum gpio_drive {
+    GPIO_DRIVE_DEFAULT,     /* Default line drive (push-pull) */
+    GPIO_DRIVE_OPEN_DRAIN,  /* Open drain */
+    GPIO_DRIVE_OPEN_SOURCE, /* Open source */
+} gpio_drive_t;
+
 typedef struct gpio_handle gpio_t;
 
 /* Primary Functions */
 gpio_t *gpio_new(void);
 int gpio_open(gpio_t *gpio, const char *path, unsigned int line, gpio_direction_t direction);
 int gpio_open_name(gpio_t *gpio, const char *path, const char *name, gpio_direction_t direction);
+int gpio_open_advanced(gpio_t *gpio, const char *path, unsigned int line, gpio_direction_t direction, gpio_bias_t bias, gpio_drive_t drive, bool inverted, const char *label);
+int gpio_open_name_advanced(gpio_t *gpio, const char *path, const char *name, gpio_direction_t direction, gpio_bias_t bias, gpio_drive_t drive, bool inverted, const char *label);
 int gpio_open_sysfs(gpio_t *gpio, unsigned int line, gpio_direction_t direction);
 int gpio_read(gpio_t *gpio, bool *value);
 int gpio_write(gpio_t *gpio, bool value);
@@ -63,15 +78,22 @@ int gpio_poll_multiple(gpio_t **gpios, size_t count, int timeout_ms, bool *gpios
 /* Getters */
 int gpio_get_direction(gpio_t *gpio, gpio_direction_t *direction);
 int gpio_get_edge(gpio_t *gpio, gpio_edge_t *edge);
+int gpio_get_bias(gpio_t *gpio, gpio_bias_t *bias);
+int gpio_get_drive(gpio_t *gpio, gpio_drive_t *drive);
+int gpio_get_inverted(gpio_t *gpio, bool *inverted);
 
 /* Setters */
 int gpio_set_direction(gpio_t *gpio, gpio_direction_t direction);
 int gpio_set_edge(gpio_t *gpio, gpio_edge_t edge);
+int gpio_set_bias(gpio_t *gpio, gpio_bias_t edge);
+int gpio_set_drive(gpio_t *gpio, gpio_drive_t drive);
+int gpio_set_inverted(gpio_t *gpio, bool inverted);
 
 /* Miscellaneous Properties */
 unsigned int gpio_line(gpio_t *gpio);
 int gpio_fd(gpio_t *gpio);
 int gpio_name(gpio_t *gpio, char *str, size_t len);
+int gpio_label(gpio_t *gpio, char *str, size_t len);
 int gpio_chip_fd(gpio_t *gpio);
 int gpio_chip_name(gpio_t *gpio, char *str, size_t len);
 int gpio_chip_label(gpio_t *gpio, char *str, size_t len);
```

The `gpio_open_advanced()` and `gpio_open_name_advanced()` are a bit of a mouthful, but they support everything the kernel's got so far for gpio-cdev, and shouldn't be too big of a problem for language bindings.

sgjava, 2020-05-19:
Cool, once you push changes I'll add to my Java bindings.

commit c0e37820 (2020-05-23): gpio: add getters and setters for bias, drive, inverted

resolves #23.
commit b6de72b2 (2020-05-27): gpio: add getters and setters for bias, drive, inverted

resolves #23.
commit e7db5145 (2020-05-28): gpio: add getters and setters for bias, drive, inverted

resolves #23.
commit 33ff2a5b (2020-05-28): gpio: add getters and setters for bias, drive, inverted

resolves #23.
closed 2020-05-29 by vsergeev