vsergeev/c-periphery #32: How to keep the pin's value

yyykkkyyy, 2020-09-11:
Hi, I really enjoy using this amazing gpio library !

BTW, when I use `gpio_write` function, it doesn't keep the value.
If I write 1 to the pin connected LED,  the LED flashes for a moment.
Is this correct behavior ?
And can I keep the value without `sleep()` in C code ?

Thanks.

vsergeev, 2020-09-11:
Hi @yyykkkyyy. It sounds like your program is terminating after the gpio_write()? If you are using character device GPIOs, e.g. with a `/dev/gpiochipN` device, you should keep in mind that the GPIOs won't retain their output state and will revert to a default state (likely an input) after being released when the program exits.

yyykkkyyy, 2020-09-11:
Thank you for your replying @vsergeev !!
Yeah, as you mentioned above, I use `/dev/gpiochipN` char device.
And I use the sample code which is on the README.

> If you are using character device GPIOs, e.g. with a /dev/gpiochipN device, you should keep in mind that the GPIOs won't retain their output state and will revert to a default state (likely an input) after being released when the program exits.

Is this behavior the specification of `/dev/gpiochipN?`
Can I retain the GPIO's state ?

vsergeev, 2020-09-11:
Yeah, that's the behavior of the Linux character device GPIO API. You can retain state after close with the sysfs GPIO API if you export the GPIO before c-periphery opens it (`gpio_open_sysfs()`), but the Linux sysfs GPIO API is deprecated and will be removed in future versions of Linux.

yyykkkyyy, 2020-09-11:
Oh, Okay. I understood the behavior !
In fact, I used the sysfs GPIO but I knew it will be deprecated. So I explore the other way.
Thank you very much @vsergeev !!!
And sorry for my poor English :-)

closed 2020-09-11 by yyykkkyyy

vsergeev, 2020-09-11:
No worries. Feel free to open an issue if you have any other questions or problems.