vsergeev/c-periphery #35: Bias compile settings

sgjava, 2020-11-05:
I had to use CFLAGS=-DPERIPHERY_GPIO_CDEV_SUPPORT -DGPIOHANDLE_REQUEST_BIAS_PULL_UP -DGPIOHANDLE_REQUEST_BIAS_PULL_DOWN -DGPIOHANDLE_REQUEST_BIAS_DISABLE or I'd get "Kernel version does not support configuring GPIO line bias".  Is that the correct way to use bias?

vsergeev, 2020-11-07:
If your kernel version / headers are older than 5.5, then they don't support GPIO line bias. You'll either need to build with a new kernel version, or c-periphery will need to add conditional compilation for that specific gpio-cdev feature. gpio-cdev was originally introduced (without line bias feature) in kernel version 4.8.

gpio-cdev has been under a lot of flux across kernel versions, which is good for new features, but adds some complexity to libraries like c-periphery downstream. There appears to be a new v2 API coming to kernel version 5.10. It's going to take a bit of effort to seamlessly support multiple kernel versions.

sgjava, 2020-11-07:
Ah, totally makes sense since I'm testing on 5.4 kernel under Armbian. I believe they stick with LTS versions only, so it will be next LTS.

sgjava, 2020-11-07:
Oh, actually, there's 5.8 mainline, I will try this now...

sgjava, 2020-11-07:
OK this works `Linux nanopiduo 5.8.16-sunxi #20.08.14 SMP Tue Oct 20 22:15:32 CEST 2020 armv7l armv7l armv7l GNU/Linux` even though the Duo doesn't have PUDs :) At least the kernel error went away.

closed 2020-11-07 by sgjava