vsergeev/c-periphery #43: Update GPIO code?

sgjava, 2021-11-26:
I noticed this while browsing the gpio.h source. You use it here https://github.com/vsergeev/c-periphery/blob/2379567960b9f72ccfb8f9db7271092612e89bdf/src/gpio.c#L991

```
/**
 * struct gpioevent_data - The actual event being pushed to userspace
 * @timestamp: best estimate of time of event occurrence, in nanoseconds
 * @id: event identifier
 *
 * Note: This struct is part of ABI v1 and is deprecated.
 * Use &struct gpio_v2_line_event instead.
 */
struct gpioevent_data {
	__u64 timestamp;
	__u32 id;
};
```

vsergeev, 2023-03-31:
v2.4.0 is now released with gpio-cdev v2 ABI support, so c-periphery will automatically build the v2 ABI implementation if the kernel headers support it.

closed 2023-03-31 by vsergeev

sgjava, 2023-03-31:
Thanks man.