vsergeev/c-periphery #44: ADC - implemented?

habakuk1979, 2022-01-01:
Hello!

have you implemented the ADC (analog inputs) in your code too?

pezi, 2022-01-05:
The Raspberry Pi and most clones computer do not have a way to read analog inputs. This functionaly is provided by additonal hats or projects like this one:
https://learn.adafruit.com/reading-a-analog-in-and-controlling-audio-volume-with-the-raspberry-pi?view=all

sgjava, 2022-01-05:
For a LDR you can just use a capacitor https://www.instructables.com/Raspberry-Pi-GPIO-Circuits-Using-an-LDR-Analogue-S and digital pins.

vsergeev, 2023-03-31:
This might be implemented in the future through the IIO sysfs interface. Similar request has been made in vsergeev/python-periphery#19.

vsergeev, 2025-03-04:
I looked into the sysfs iio interfaces a bit more, and wasn't able to find a consistent API for ADCs. If someone has a suggestion for common sysfs attributes, we could discuss a possible ADC module. Closing in the meanwhile.

closed 2025-03-04 by vsergeev