vsergeev/c-periphery #54: Unknown CMake command "check_source_compiles" on Debian Bullseye

Thi3rryzz, 2024-03-12:
Hi all,

I'm trying to build this source on a debian bullseye machine without success.
I'm facing the following CMake issue:

`CMake Error at src/c-periphery/CMakeLists.txt:12 (check_source_compiles): Unknown CMake command "check_source_compiles".`

If I'm not mistaken, this command was only introduced in CMake version 3.19, where Debian bullseye ships with 3.18.4 by default.
Any suggestions on how to go about this? (preferrably without utilizing bullseye-backports version of CMake?)

Thank in advance!
Thierry
Also, the cmake_minimum_required version is currently set at 2.6, which should be upped I think?

LenBonJJ, 2025-02-19:
the same question，do you solve？

Thi3rryzz, 2025-02-20:
Nope, still not resolved.

commit b4ca16e6 (2025-02-27): cmake: update minimum cmake version required

for CheckSourceCompiles module.

resolves #54.
commit 41d04475 (2025-02-27): cmake: update minimum cmake version required

for CheckSourceCompiles module.

resolves #54.
commit 629f63ea (2025-02-27): cmake: update minimum cmake version required

for CheckSourceCompiles module.

resolves #54.
commit fa95c4dd (2025-02-28): cmake: update minimum cmake version required

for CheckSourceCompiles module.

resolves #54.
closed 2025-03-01 by vsergeev