vsergeev/c-periphery #56: Makefile incorrectly set the flags for character device support

fell-z, 2024-09-03:
Hey.

I had some issues using the lib via [lua-periphery](https://github.com/vsergeev/lua-periphery). After some digging of my own, I discovered that 'c-periphery' wasn't building with character device support, so I took a look at the Makefile and found a problem in the following line.

```makefile
GPIO_CDEV_SUPPORT = $(if $(filter 1,$(GPIO_CDEV_V2_SUPPORT)),2,$(if $(filter 1,$(GPIO_CDEV_V1_SUPPORT)),1,0))
``` 

In those conditions, `filter 1` should be `filter 0` to work correctly.

reeett, 2024-09-06:
Hi @fell-z,
had the same problem writing a Makefile of lua-periphery for OpenWRT. Your suggestion fixed it for me too.

Thank you

---

Patchfile for OpenWRT 23.05.4
```
Index: lua-periphery-2.4.2/c-periphery/Makefile
===================================================================
--- lua-periphery-2.4.2.orig/c-periphery/Makefile
+++ lua-periphery-2.4.2/c-periphery/Makefile
@@ -12,7 +12,7 @@ OBJECTS = $(patsubst $(SRCDIR)/%.c,$(OBJ
 
 GPIO_CDEV_V1_SUPPORT := $(shell ! echo -e "#include <linux/gpio.h>\n#ifndef GPIO_GET_LINEEVENT_IOCTL\n#error\n#endif" | $(CC) -E - >/dev/null 2>&1; echo $$?)
 GPIO_CDEV_V2_SUPPORT := $(shell ! echo -e "#include <linux/gpio.h>\nint main(void) { GPIO_V2_LINE_FLAG_EVENT_CLOCK_REALTIME; return 0; }" | $(CC) -x c - >/dev/null 2>&1; echo $$?)
-GPIO_CDEV_SUPPORT = $(if $(filter 1,$(GPIO_CDEV_V2_SUPPORT)),2,$(if $(filter 1,$(GPIO_CDEV_V1_SUPPORT)),1,0))
+GPIO_CDEV_SUPPORT = $(if $(filter 0,$(GPIO_CDEV_V2_SUPPORT)),2,$(if $(filter 0,$(GPIO_CDEV_V1_SUPPORT)),1,0))
 
 COMMIT_ID := $(shell git describe --abbrev --always --tags --dirty 2>/dev/null || echo "")
 

```

vsergeev, 2025-02-27:
If you're seeing `0` for `GPIO_CDEV_V1_SUPPORT` or `GPIO_CDEV_V2_SUPPORT`, that means there was an error in building the support test, so there might be a problem with those tests on some platforms.

`$(shell ! ...; echo $$?)` inverts the status code, so if the support test succeeded, i.e. exited 0, then the variable will evaluate to 1.

`$(if $(filter 1,$(GPIO_CDEV_V2_SUPPORT)),2,$(if $(filter 1,$(GPIO_CDEV_V1_SUPPORT)),1,0))` evaluates to 2 if `GPIO_CDEV_V2_SUPPORT` is 1, to 1 if `GPIO_CDEV_V1_SUPPORT` is 1, and otherwise 0.

Changing it to `$(if $(filter 0 ...` will make it succeed and indicate v2 support, but only for a failing a support test...

Probably the most useful info would be the output of `echo -e "#include <linux/gpio.h>\n#ifndef GPIO_GET_LINEEVENT_IOCTL\n#error\n#endif" | $(CC) -E - >/dev/null 2>&1` on one of these platforms where this is failing.

commit 2077ae48 (2025-02-28): make: use printf instead of echo for gpio cdev support tests

some shells (e.g. dash) don't support the -e switch on echo for
interpreting escape sequences. printf is a more portable alternative.

resolves #56.
commit d7d426ea (2025-02-28): make: use system printf for gpio cdev support tests

some shells (e.g. dash) don't support the `-e` switch with their
built-in echo for interpreting escape sequences. also, the `-e` switch
is technically not POSIX compliant. so use system (e.g. coreutils,
busybox) printf instead.

resolves #56.
closed 2025-03-01 by vsergeev